COMPASS-DOE/TEMPEST #4: Field Work

stephpenn1, 2021-01-04:


stephpenn1, 2021-01-12:
@bpbond can you grab a power supply cable from 1207 for the AC to DC power adapter? See specs below...

- Part Number 9981-140 Power Supply Cable
- 4 pin (F-F) 2m

My guess is it's either yellow like the photo below or grey.

### From the manual: 
> The AC to DC power supply includes a 2 m power cord that connects the power supply to the LI-8150 Multiplexer (part number 9981-140). Connect the power cord between the power supply and multiplexer before plugging the power supply into a power outlet.

[image]

stephpenn1, 2021-01-14:
**Note to talk to licor about:** When I pulled out a longterm chamber from the salt plot, the inside of the chamber was covered in sandy soil sensors included. I assume the LT chambers are built for extended outdoor use... but I'm concerned it could be harmful to the measurement. 

[image]

bpbond, 2021-01-14:
Ok, Norbert is started for test. It's measuring every two hours, started 1:05 PM today, logging to internal flash ￼(almost no free space on either flash card).

closed 2021-04-07 by stephpenn1