COMPASS-DOE/TEMPEST #7: Repository size and other issues

bpbond, 2021-02-25:
Hello-

This repository is already over 2.5 GB, which is _really_ large. GitHub support will start emailing you at around 5 GB I think.

Also, it seems to have a full copy of of the [premis-ghg](https://github.com/PNNL-PREMIS/PREMIS-ghg) repo embedded within it, in `Soil Transplant Exp/`? Did you mean to do this? That accounts for about two-thirds of the size, and seems...odd. But maybe I'm not understanding the intent.

Anyway, just wanted to give you a heads-up on the size issue, and clarify the structuring decisions (including making TEMPESTX a sub-folder versus its own repo). But these are your decisions, not trying to overly second-guess. Thanks!

bpbond, 2021-02-25:
P.S. that's a wonderful README 👏

ahopple, 2021-02-25:
Hey!

Yikes! Thank you for the heads up. I had originally added both TEMPESTX and the Transplant Exp to the TEMPEST repo (when I was a Github baby, but now I'm a GitHub toddler). Completely agree that it doesn't make sense to have them embedded in the TEMPEST repo. I'll remove the premis-ghg repo and make TEMPESTX its own repo under COMPASS - does that seem like a good way to structure the projects?

stephpenn1, 2021-02-25:
@ahopple @bpbond I removed premis-ghg and any duplicate data in my `combine-repos` branch

stephpenn1, 2021-02-25:
@ahopple Yes I think since TEMPESTX will have it's own data files separate from the overall experiment, it will be good to create a separate repo

ahopple, 2021-02-26:
TEMPESTX and Transplant Projects have been removed.

closed 2021-02-26 by ahopple