COMPASS-DOE/TEMPEST #11: FAGR inventory tags

stephpenn1, 2021-03-01:
Beeches are hungry by nature and like to eat up tags, some (e.g. tag 1829) are hammered in too close and need to be pulled out a bit.. let's do this when we do 2021 inventory

stephpenn1, 2021-03-01:
### Also when we do inventory:
- check species of F12 (1563) it seems to be incorrectly labeled on our sheets

stephpenn1, 2021-03-05:
@bpbond For Tuesday, can you grab the inventory nails and tags and also the blue roll of sapflow tree putty in 1207 for @aestearns01

Alice, we will go around and pull out the nails in the Beeches during our inventory so you can take that off your to do list. However we will leave some spare nails and the putty in the PNNL cabinet

bpbond, 2021-03-05:
Will do @stephpenn1

aestearns01, 2021-03-06:
Thanks Stephanie! I noticed that there is currently a roll of putty in the
tool bag in the PNNL cabinet, but it couldn't hurt to have more.

PR #22 (2021-03-23): 2021 inventory with updated format
- updated format using ForestGEO conventions (For `Status_Year`: LI = living, DS = dead, standing, DC = dead, down). will be better for computing & analysis
- 2021 values input from 3/9/2021
- Survey of what trees are out of plot are marked with an "Out of plot" in the `Notes` column

Questions: 
1. Should we switch the "Out-west" and "Out-east" to simply "West" and "East"?
2. For 1812 and 1813 in the Salt plot, it looks like you (@bpbond) switched the numbers in the data sheet. Does this mean the values were for the wrong tag or just that the rows are out of [... cut]
closed 2021-03-23 by stephpenn1