COMPASS-DOE/TEMPEST #13: TEROS pipeline review request

ahopple, 2021-03-05:
Hi @bpbond 

Could you please take a look at the revised "TEROS12_Pipline.R" script? It's located here: TEMPEST/Data/Soil VWC, T, and EC (TEROS12)/

I really tried to embrace the "if you're copying and pasting, you're doing something wrong" principle :-) The script collapses all of the TEROS data into a single data frame. I did have trouble reshaping each variable into its own column and tagged you with "BBL" on line 65 with a question. 

I feel like this is a HUGE coding step forward for me and really appreciate all of your guidance and feedback. 

Looking forward to your thoughts!

bpbond, 2021-03-05:
Hi @ahopple looks GREAT!  👏   I'll open a PR but a few notes here:

* You deleted the RStudio project file; any particular reason? If not let's leave it please. Restoring.
* You never ever ever :) want to do `setwd()` at the beginning of your script. It makes scripts fragile and unusable by anyone else; see [here](https://rmd4sci.njtierney.com/workflow.html) and [here](https://github.com/jennybc/here_here#admitting-you-have-a-problem).
* I fixed the problem (initial blank line) in `TEROS11_10202020.txt`
* I added a pattern-match to the `list.files()` call so it only looks for TEROS files, not e.g. README files
* Simplified overly-complex `apply` calls 
* Changed from `x1`, `x2`, etc. to more informative names
* You like your comments to have no space at the beginning, while I do. It's a war 😄 
* I moved the `na.omit()` much earlier in the pipeline—there's a lot of data here, so we want to remove missing rows as early as possible
* Use `tibble` instead of `data.frame` for data—much easier to print
* The `spread` wasn't working because there are duplicate rows; fixed
* Changed plots to give each sensor its own line
* Looks like we have some sensors with no matching plot

bpbond, 2021-03-05:
I just noticed the timestamp of this issue. Hahaha hi there night owl  🦉

ahopple, 2021-03-05:
Thank you @bpbond !! Yes, I was a total night owl 😄 It has been difficult to be selfish with my time the last two weeks given the field work load, so I just had to find time to work on my coding. It's also very hard to stop once you get going...

I added a few responses in bold below.

- You deleted the RStudio project file; any particular reason? If not let's leave it please. Restoring.
**Sorry about that. I was cleaning up my old files and saw that it was empty, so I deleted it. Forgot that we had just created it :-/**

- You never ever ever :) want to do setwd() at the beginning of your script. It makes scripts fragile and unusable by anyone else; see here and here.
**Got it**

- I fixed the problem (initial blank line) in TEROS11_10202020.txt 👍 

- I added a pattern-match to the list.files() call so it only looks for TEROS files, not e.g. README files 👍 

- Simplified overly-complex apply calls 👍 

- Changed from x1, x2, etc. to more informative names 👍 

- You like your comments to have no space at the beginning, while I do. It's a war 😄
**haha! just like folks that write proposal/papers with two spaces between sentences - you're probably one of those too 😄**

- I moved the na.omit() much earlier in the pipeline—there's a lot of data here, so we want to remove missing rows as early as possible 👍 

- I suggest using tibble instead of data.frame for data—much easier to print 👍

- The spread wasn't working because there are duplicate rows; fixed 👍

- Changed plots to give each sensor its own line 👍

- Looks like we have some sensors with no matching plot
**I'll take a look at this. There are many data table channels that will have no plot, but all sensors should have a plot**

PR #14 (2021-03-05): Suggested tweaks and streamlining
See #13 for full list of changes.

Closes #13
closed 2021-03-05 by ahopple