COMPASS-DOE/TEMPEST #20: Tree inventory QA/QC analysis script

bpbond, 2021-03-10:
Hi @10aDing thanks for being willing to work on this!

The `inventory.csv` file describes, for each tree in the TEMPEST plots, its "DBH" (diameter) in 2019 and 2020 (and soon 2021):

Plot | Section | Tag | DBH_2019 | Date_2019 | Species_code | Status_2019 | Sapflux | DBH_2020 | Date_2020 | Status_2020 | Notes | JS_codes
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Control | East | 1301 | 10.7 | 3/9/19 | COFL2 | Yes |   | 10.7 | 3/10/20 |   |   |  
Control | East | 1302 | 11.7 | 3/9/19 | ACRU | Yes |   | 11.7 | 3/10/20 |   |   |  
Control | East | 1303 | 18.7 | 3/9/19 | ACRU | Yes |   | 18.9 | 3/10/20 |   |   |  
Control | East | 1304 | 51.8 | 3/9/19 | LIST2 | Yes |   | 52.2 | 3/10/20 |   |   |  

We would like a script that does qa/qc (error checking and visualization), along later with some computation on the data. Initially:
* Create a new branch for your work
* Create a new R Markdown file in the `R_scripts/` folder 
* Read in the above dataset
* Produce a summary table (probably using `dplyr` functions, but up to you) showing, by plot and species, number of trees 
* Identify any trees whose diameter changed more than ±0.5 cm between 2019 and 2020
* Make a nice graph of the data! Think about how we might visualize the above information

bpbond, 2021-03-11:
Also pinging @stephpenn1 @ahopple to make sure they're aware of this work.

bpbond, 2021-03-18:
Nice start! https://github.com/10aDing/TEMPEST/blob/QA/QC/R_scripts/Colin_QAQC.Rmd

One important thing we'd like is, when we add a new year of data (as @stephpenn1 is about to do), we don't need to rewrite the script.

I think we will need to _reshape_ the data frame into a 'long' form: instead of columns `Tag`, `DBH_2019`, and `DBH_2020`, we'll have `Tag`, `Year`, and `DBH`.

For this you'll want to 
* use `tidyr::pivot_longer()` to reshape the data;
* use `tidyr::separate()` to separate the `DBH_2019` (e.g.) labels into two columns.

```R
dat <- data.frame(Tag = 1:2, DBH_2019 = 2:3, DBH_2020 = 3:4)
library(tidyr)
library(dplyr)
dat %>% 
  pivot_longer(starts_with("DBH"), values_to = "DBH") %>% 
  separate(name, into = c("junk", "year")) %>% 
  select(-junk) ->
  dbh_data
```

You can do this for DBH and for `Status_2019`, etc. And then use `left_join()` to join `dbh_data` and `status_data` together.

After this, please produce a table (and a figure?) showing number of trees by plot and year. (And species?)

After this we will move into some real QA/QC steps, for example, identifying zombie trees that were recorded dead but then came back to life   🧟

bpbond, 2021-03-18:
Quick note:
```R
inventory <- inventory %>%
  mutate(Sig_growth_2020 = case_when(
    DBH_2020 >= DBH_2019 + 0.5 ~ TRUE,
    DBH_2020 <= DBH_2019 - 0.5 ~ TRUE,
    TRUE ~ FALSE
  ))
```
could be more simply written as 
```R
inventory$Sig_growth_2020 <- abs(DBH_2020 - DBH_2019) > 0.5
```

bpbond, 2021-03-25:
Nice job @10aDing ! Looks great.

I think our next step is to incorporate the new tree inventory dataset that @stephpenn1 recently committed. We can do this by following [these steps](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/syncing-a-fork) to update _your_ repository from the"upstream" one. (In other words, we're updating your fork from the original.)

This operation will change the structure of the inventory dataset a bit, so re-run and test your script on the new data. Something will probably have broken :)

After that, we'd like the Rmd file to produce some nice-looking summary tables and figures:
* Number of live and dead stems, by plot and year
* Basal area by species, plot, and year probably. I'm not sure the best way to visualize this, and would suggest exploring a bit. We'd like to see change in BA over time. Basal area is defined as the cross-section area of stems per unit ground area (in m2/ha). DBH is diameter, and from that you can compute an area for each tree; plot area is 2000 m2.
* Number of standing versus down stems, by year. In the `tree_inventory/` folder there's now a README.md file that describes how the "Status" column encoding works.
* Make nice-looking tables! There are a number of ways to do this, e.g. https://rmarkdown.rstudio.com/lesson-7.html

Into bonus material (don't focus on this coming week, and we will expand this list in consultation with @stephpenn1 and @ahopple ):

We will also want to have some QC reports that warn us of unusual and/or perhaps wrong things:
* Illegal status entries (that don't correspond to the guide)
* Negative DBH
* We will come up with more...

bpbond, 2021-03-25:
P.S. once we get through the fork update and first steps above, i.e. the script is producing a few (not a lot!) useful figures and tables, we will open a PR to the main repo and get feedback from @stephpenn1 and @ahopple .

bpbond, 2021-04-02:
Nice work!

1. You're going to want to filter for `In_plot`
2. For the num_trees tables, let's filter for live trees only and put one year per column
2. Mortality table (Alive versus dead by plot and year) should also be a ggplot!

Let's think about mistakes this script can identify:

1. Dead in a year but then alive in any succeeding year
2. Down in a year but then standing in any succeeding year
3. If Status is _not_ alive, dead, or dead-down, or if it's blank
4. DBH should always be >= 1

For each of these, the script should (in a new "Errors" section, use big type!) display the problem entries, if any exist.

bpbond, 2021-04-02:
Also — for basal area, try using a `geom_area()` (stacked area plot).

bpbond, 2021-04-02:
One more thing about basal - please report total basal area for each plot. (This should be a number in the 20-40 range.)

bpbond, 2021-04-08:
From meeting with @stephpenn1 and @ahopple -

* **Distribution of diameters by plot (and species?)**
* Sig_growth should be calculated, not fixed, based e.g. on 95% extremes of growth
* Incorrect status - don't flag ones that enter the dataset
* Spatial plotting (see #33 )
* **Visualize mortality by year**
* Estimate crown size via allometry?
* Error check—species code mistakes?
* **List not available species**
* Date error check?
* **Species distribution, counts and BA, by plot**

bpbond, 2021-04-19:
Next (and really final) steps-

* Once @stephpenn1 checks and merges #34 , the tree inventory dataset will have a new field called "Grid"
* This is a two-character code that provides the grid location for each tree within the plot. Axes are A-J and 1-8, so for example "D3" says the tree is in the  {3, 3} square ("like Battleship"); "A7" is the {1, 7} square 
* The final step for your script will be to visualize the spatial distribution of trees in each, coloring by species
* You'll want to split the Grid field into "x" and "y" for plotting, and use `geom_jitter()` to space out points in grid cells

That sound good?

stephpenn1, 2021-04-20:
Everything is merged, please proceed :)

PR #37 (2021-06-10): Add tree inventory data QA/QC Script
Add the QA/QC script specified in Issue #20 to create a long table, graph data, and identify errors.

Closes #20
closed 2021-06-10 by bpbond