COMPASS-DOE/TEMPEST #23: TEROS Pipeline - review request

ahopple, 2021-03-19:
Hi @bpbond
 
Could you review the updates to the TEROS pipeline? I've tagged you with BBL where I had questions. Thank you!

bpbond, 2021-03-19:
Notes
* We're going to need to think about data sizes. The `teros_data` object is already 130 MB. I did some work to trim it down.
* NA plot rows—this is happening just because the data files have `Data_Logger_ID` / `Data_Table_ID` combinations that aren't in the map file:

[image]

Related question for @ahopple : why do we have both "TEMPEST_TEROS_12_location_and_ID.csv" (in `Design_Map/`) and "TEMPEST_TEROS_Network_Location&ID.csv" (in `Raw_Data/`)?

bpbond, 2021-03-20:
Re above: for example cataloguer ID 41 doesn't have any entries for data table IDs 12, 13, and 17-22.

bpbond, 2021-03-20:
```
# BBL what are your thoughts on data cleaning? I'm hopeful that we won't have to deal with these
# large value jumps often now that we have fixed most of the TEROS network issues. For now, I am identifying
# the faulty sensors as those with values 2 ST DEV outside of the mean and removing them. We might want to use a
# moving average or something similar to identify sensor errors down the road.
```
I would suggest we produce a report for how many values are removed, by sensor and day perhaps.

For filtering, we don't want to do this:
```r
teros_data2 %>%
  filter(...
```
...because the whole row will get removed if even one of TSOIL, VWC, or EC is bad. Also we don't want to use a standard deviation-based criterion, as that's very sensitive to outliers; using the mean absolute deviation is [better](https://core.ac.uk/download/pdf/206095228.pdf).

bpbond, 2021-03-20:
@ahopple opening PR with some suggested changes.

bpbond, 2021-03-20:
>Calculating daily averages - or do we want to keep the 15-minute data, BBL?

Good question. For QA/QC I think daily data are generally going to be much more useful, although obviously we want to keep the 15-minute data available for anyone who wants it.

bpbond, 2021-03-20:
@ahopple I have changed the raw data plotting in the PR to group by datalogger—thoughts?

[image]

closed 2021-06-07 by bpbond