COMPASS-DOE/TEMPEST #32: Licor analyzers

bpbond, 2021-04-06:
From call with Kristen Minish.

* Loaner of 7820 (N2O) + new multiplexer + hopefully 4 LTCs
* Plumb desiccant inline before flooding
* Can mount analyzers on shipping pallets, cinderblocks, etc.
* Trace gas analyzers (and power supplies) are not weatherized (unlike multiplexer)
* Licor uses a doghouse to protect them!
* Methane analyzer - the two vents are not weatherized - they need 1-2" clearance to cool instrument
* Licor website has options for weather traps - https://licor.com/env/support//LI-7810/topics/avoiding-water-ingress.html
* Take a hard look for any condensation in lines from LTC -> multiplexer -> analyzer
* Definitely can loan analyzer, she will check about multiplexer & let know within a week or two
* Have analyzer to us by beginning of June (maybe sooner)

[image]

bpbond, 2021-04-07:
>We should track instrument serial numbers for TEMPEST

>Maybe we have a “licor_installation.csv” file that has date, plot, nickname, serial number

stephpenn1, 2021-09-28:
Closing because test event and loaners are done :)

closed 2021-09-28 by stephpenn1