COMPASS-DOE/TEMPEST #35: SP Rest of April

stephpenn1, 2021-04-19:
@bpbond 

#### TEMPEST - data
- [ ] Develop plan for the data processing of each data stream (including an inventory of what scrips have already been written)
- [ ] Check sapflow values now that leaves are out
- [ ] Update READMEs

#### TEMPEST - field
- Can you let me know of anything you'd like me to do / check on

#### Syntheses 
- Let's talk next steps on Friday AM if you're available

bpbond, 2021-04-19:
Sounds good SP—thank you.

bpbond, 2021-04-24:
Tuesday 4/27

- O.o.o. notifying 
- Interns
- Anya meeting
- TEMPEST scripting 
- Data synthesis

bpbond, 2021-05-03:
Monday 5/3

* Week plans? Please create a new issue
* Specific next steps for data processing? Progress?
* Meta-analysis - incorporating 2018-2020 data into script
* IRGA checking and downloading - how to handle
* Other?

closed 2021-05-11 by stephpenn1