COMPASS-DOE/TEMPEST #45: Update field laptop

bpbond, 2021-06-04:
@stephpenn1 can you take WE34259 and put it onto the PNNL network, or take it into the office if needed, and work with Sean to get any needed updates applied? Right now it's saying that Windows needs to be activated. Assigning this to you as you're the primary (official) user. Thanks.

bpbond, 2021-06-04:
I will set up milestones and use those to assign due dates, if that works.

bpbond, 2021-06-12:
@stephpenn1 I left your laptop in 1207 with two SIM cards on top of it: from Norbert (in TEMPEST control currently) and the unit currently in 1207 (I think [Stuart](https://www.getbeautified.com/minion-names/#18_Stuart_8211_Minion_Names)). Can you download anything on these cards and reformat them as part of updating the laptop? Thank you

stephpenn1, 2021-06-14:
Yup thanks!

stephpenn1, 2021-06-15:
Laptop is updated and was put on the pnnl network, dropped back off at GCReW in the top shelf of the PNNL cabinet

closed 2021-06-15 by stephpenn1

bpbond, 2021-06-15:
Hallelujah—thanks! I've been getting so tired of the warning messages :)