COMPASS-DOE/TEMPEST #49: Sapflow QA/QC

bpbond, 2021-06-05:
* Provide feedback / merge BBL PRs
* Work with A & A to modify report based on their feedback

Also includes #48 and #43

stephpenn1, 2021-06-08:
- [ ] Add grid square to table

bpbond, 2021-06-08:
Good idea!

stephpenn1, 2021-06-20:
From 6/16 meeting:
- compare by neighboring days
- do we want to have something that automatically displays data and stats based on the selected time frame?
- compare by logger

bpbond, 2021-06-24:
Proposed outline for QA/QC script:
 
TEMPEST QA/QC Script for Field Technician Use:

## Section 1: Realtime data overview
* Display of daily data per sensor – line graph
* Spatial display of daily data
* Also useful for identifying “nodal” issues when troubleshooting

## Section 2: Immediate action tests
* Data absence check
* Data range check
* Assess errors by data logger
* Flagged sensors require immediate action. Follow below steps:
* Check wiring/distribution block connections
* Check power supply
* Check installation site
* Check address
* Check data logger, junction boxes
* Others?

## Section 3: Sensor lifetime tests
* Data trending test
* Include sensor replacement date
* Change in error number test 
* Requires pulling in longer term data
* Flagged sensors may require replacement. Follow below steps:
* Is the sensor trending? If so, for how long?
* Has there been a big jump in the number of errors reported?
* Has the long term behavior of the sensor changed?
* If the above conditions are met, replace sensor.

stephpenn1, 2021-06-25:
thanks bbl

closed 2021-07-12 by stephpenn1