COMPASS-DOE/TEMPEST #58: SP Week of 6/28

stephpenn1, 2021-06-28:
Finish #49 
- [ ] Merge current PR #57 
- [ ] Add install date to summary table
- [ ] Add graph of deeper sensors and send to Alice
- [ ] Create spatial map using grid column (is Alice doing this?)

Finish #50 
- [ ] To fill in

... more following meeting

stephpenn1, 2021-07-06:
Hi @bpbond I am behind on this task list because of the busy-ness of last week. I will be finishing the sapflow tasks today and the teros process step and would like to meet tomorrow if possible to go over everything.

bpbond, 2021-07-06:
@stephpenn1 Ah, funny, I was just coming here to check this list. OK, sounds good, thank you. Could you also update our "SCP task overview" google sheet please?

Tomorrow...hmm, I won't have much time. If we meet with Adam at 9, maybe you and I could touch base immediately after—would that work?

closed 2021-07-06 by stephpenn1