COMPASS-DOE/TEMPEST #115: tree gas flux

kendalynnm, 2024-04-23:
Hi TEMPEST data team!
Who's excited for TEMPEST III? I know I am!
In preparation, I'm hoping to look at tree gas flux data from all previous events. Seems like it is missing from the repo, is that intentional?

Here is where the raw data lives in the gdrive: https://drive.google.com/drive/folders/13uyYcX7DcjTMkdLPoPpuu1Y8KMQV4RK_

I can spend some time working on a data processing pipeline, but want to make sure I'm starting in the right place.

kendalynnm, 2024-04-29:
> Hi TEMPEST data team! Who's excited for TEMPEST III? I know I am! In preparation, I'm hoping to look at tree gas flux data from all previous events. Seems like it is missing from the repo, is that [...]

Requesting write access to address the topic above @bpbond or @stephpenn1 , thanks in advance!

stephpenn1, 2024-04-29:
Hi Kendal, just gave you write access to this repo.

The google drive link you sent gives me a 404 error, are you planning to move those raw data into this repo?

If so, I'll also point out you can use the google drive R package to pull files in your script without having to put them in two places.